cdsteinkuehler/beaglebone-universal-io #36: loading firmware

thisboyiscrazy, 2015-08-20:
I'm am getting:

[  281.131733] bone_capemgr bone_capemgr: part_number 'cape-universala', version 'N/A'
[  281.131779] bone_capemgr bone_capemgr: slot #4: override
[  281.131801] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[  281.131824] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,cape-universala'
[  281.153568] __of_adjust_tree_phandle_references: Illegal property (size) 'fixup' @/__local_fixups__
[  281.163122] bone_capemgr bone_capemgr: slot #4: Failed to resolve tree

cdsteinkuehler, 2015-08-22:
Details, please!

What kernel version?  What image did you start with?  What's in your slots file?  What cape(s) do you have installed?

thisboyiscrazy, 2015-08-23:
4.1.5-ti-r10, ubuntu, none, I was just trying to switch all P8 to GPIO.

thisboyiscrazy, 2015-08-23:
And I'm not sure with the slots file, I do not know how to check.

cdsteinkuehler, 2015-08-23:
OK, the 4.1.x TI kernels are new and while they support device-tree overlays I haven't worked with the newer kernels yet (I need the Xenomai patches to run Machinekit).  Robert Nelson has made the updates to the universal overlay and config-pin to support the new kernels, but apparently something's borked.  I'll try to look into this, but don't have much time at the moment to experiment, 

I suggest you report the error and ask for help on the BeagleBoard Google group.  Be sure to list exactly what you did (ie: how you were trying to load the overlay) which resulted in the error.

thisboyiscrazy, 2015-09-01:
On debain I am getting:

```
[  828.540206] bone_capemgr bone_capemgr: slot #7: Failed to resolve tree
[ 1529.676981] bone_capemgr bone_capemgr: part_number 'cape-universaln', version 'N/A'
[ 1529.677066] bone_capemgr bone_capemgr: slot #8: override
[ 1529.677113] bone_capemgr bone_capemgr: Using override eeprom data at slot 8
[ 1529.677166] bone_capemgr bone_capemgr: slot #8: 'Override Board Name,00A0,Override Manuf,cape-universaln'
[ 1529.749882] of_resolve_phandles: Could not find symbol 'gpio4'
[ 1529.756244] bone_capemgr bone_capemgr: slot #8: Failed to resolve tree
```