cdsteinkuehler/beaglebone-universal-io #37: Loading cape-univ-audio and univ-hdmi

Babi-Wan, 2015-12-09:
The "state" directory is not being created when I load both capes. Only cape-universal creates this specific directory. What could have happened?

Babi-Wan, 2015-12-09:
Can I load, someway, only cape-universala? Because that's what I really need.  Hdmi and audio already have been disabled.

cdsteinkuehler, 2015-12-09:
This is a known problem with the kernel module helper used, which only exports details from the first cape loaded (see issues #29 and #30).  That's why the capeuniversala was created, which combines all three capes (universal, emmc, and hdmi).  If you still need emmc enabled, you'll need a combination of universal+hdmi, which hasn't been crafted yet.

Babi-Wan, 2015-12-09:
Got it! So I guess I'll try to craft this combination and see if works. Thanks for your help.

cdsteinkuehler, 2015-12-09:
If you get it working, send a pull request!  Otherwise, holler if you need help.

Babi-Wan, 2015-12-09:
Of course! I'll keep in touch.

Babi-Wan, 2015-12-28:
Ok, I'm trying to craft the cape, but then i've got this problem when I do the following steps:

1) load the cape: echo cape-universalb > /sys/devices/bone_capemgr.9/slots
Ok, no problem.

2) configure pin P9.14 to pwm mode: echo pwm > /sys/devices/ocp.3/P9_14_pinmux.49/state
Then I got the error "bash: echo: write error: Invalid argument" when I hit enter for the first time. The second time I hit, goes fine. But this error does not occur with cape-universaln, for example. And I believe this is the reason that makes the kernel incapable of recognizing the multiplexation.

How can I send my .dts file to you?

cdsteinkuehler, 2015-12-28:
On 12/28/2015 8:36 AM, Bárbara Brandão wrote:
> 
> How can I send my .dts file to you?

The best way is to check it in to a github clone of my repository and
send me a link.  That way, you can send a pull-request once you get it
working.  :)  Otherwise, just email it to me (charles <at>
steinkuehler.net).

I suggest after loading your new cape, you poke around in the system
log files and examine the sysfs entries for the pinmux helper before
and after trying to switch the mode to pwm.  That may help turn up
what's going wrong.

-- 
Charles Steinkuehler
charles@steinkuehler.net