cdsteinkuehler/beaglebone-universal-io #40: changing SPI1 D0 as output

rifotu, 2016-08-23:
Hello,

I don't see a way to configure SPI1 D0 as output (MOSI). Is there a way to change it? If not, can you please tell me where I should change in the source code to swap MOSI and MISO (D1 and D0)

cdsteinkuehler, 2016-08-24:
On 8/23/2016 5:00 PM, rifotu wrote:
> Hello,
> 
> I don't see a way to configure SPI1 D0 as output (MOSI). Is there a way to 
> change it? If not, can you please tell me where I should change in the source 
> code to swap MOSI and MISO (D1 and D0)

I don't use the SPI and am not sure if there is a way to swap these
pins.  If there is a way, it's probably via device tree settings.

You should ask on the BeagleBoard Google group and/or dig through the
kernel documentation and SPI driver source code.

-- 
Charles Steinkuehler
charles@steinkuehler.net

rifotu, 2016-08-24:
Hello,

Beagleboard supports changing D0 and D1 as MOSI or MISO.  There are overlay files for choosing which one is MOSI in the below link.
http://elinux.org/BeagleBone_Black_Enable_SPIDEV    -> [example overlay](http://elinux.org/images/f/f3/BB-SPI1-SWP-01-00A0.txt)

The keyword is  ti,pindir-d0-out-d1-in
https://github.com/beagleboard/linux/blob/4.4/Documentation/devicetree/bindings/spi/omap-spi.txt#L9

I just wasn't able to find where to change in universal-io source code to have this. Can you please guide me where to change to have that configuration support

thanks a lot
rifo

cdsteinkuehler, 2016-08-24:
On 8/24/2016 11:45 AM, rifotu wrote:
> 
> I just wasn't able to find where to change in universal-io source code to have 
> this. Can you please guide me where to change to have that configuration support

You will need to create a custom version of the universal overlay to
get this to work.  Remember to rename it or you will always load the
version built into the kernel.

You need to change the spi device entry and add the
ti,pindir-d0-out-d1-in line:
https://github.com/cdsteinkuehler/beaglebone-universal-io/blob/master/cape-universal-00A0.dts#L1667-L1698

You may also need to change the spi mode pinmux values for the two pins:

https://github.com/cdsteinkuehler/beaglebone-universal-io/blob/master/cape-universal-00A0.dts#L643

https://github.com/cdsteinkuehler/beaglebone-universal-io/blob/master/cape-universal-00A0.dts#L661

-- 
Charles Steinkuehler
charles@steinkuehler.net

rifotu, 2016-08-25:
thanks!
I will try them out and report back.

rowsail, 2019-03-06:
Thinking about it - I am going definitely going to change the PCB - so do not worry about that extra hassle of swapping D0 and D1.