cdsteinkuehler/beaglebone-universal-io #41: Extending wiki - how to automate "config-pin P9.28 spi" command

rifotu, 2016-08-25:
Hello,

Can we add how to automate the config-pin commands . I have tried using 
/etc/rc.local
Initd
Monit
But for some reason I wasnt able to have the system work reliably. ( I am using the latest  debian wheezy image)

Can you please add this info to wiki?
Thanks a lot
Rifo

rifotu, 2016-08-26:
Hello,

I have done some more trials. It seems that I need to insert a delay of 7 seconds in the application itself before issuing the "config-pin P9.30 spi"  command from the application itself.
without the delay, P9.30 pin stays in default mode.

mvduin, 2016-09-18:
To reduce boot time, systemd (the init system used on beaglebone debian images) starts services in parallel as much as allowed by the ordering constraints specified by those services.

To ensure you can use config-pin you need to add a constraint ordering your service after the one which loads cape-universal, which is generic-board-startup.service.

A minimal service file for doing one-time setup would be something like

```desktop
[Unit]
Description=insert description here

After=generic-board-startup.service

[Install]
WantedBy=multi-user.target

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/path/to/executable and arguments
```

rifotu, 2017-02-03:
great! thanks a lot for your help

closed 2017-02-03 by rifotu