cdsteinkuehler/beaglebone-universal-io #43: Could not find symbol 'gpio4'

Biffco, 2016-09-29:
kernel version - uname -r: 4.1.33-bone24-00149-gea9e337

command:  # echo cape-universaln > /sys/devices/platform/bone_capemgr/slots
result:
[ 2742.848488] bone_capemgr bone_capemgr: part_number 'cape-universaln', version 'N/A'
[ 2742.856294] bone_capemgr bone_capemgr: slot #7: override
[ 2742.861745] bone_capemgr bone_capemgr: Using override eeprom data at slot 7
[ 2742.868819] bone_capemgr bone_capemgr: slot #7: 'Override Board Name,00A0,Override Manuf,cape-universaln'
[ 2742.928609] of_resolve_phandles: Could not find symbol 'gpio4'
[ 2742.934584] bone_capemgr bone_capemgr: slot #7: Failed to resolve tree
bash: echo: write error: Invalid argument

command: # cat  /sys/devices/platform/bone_capemgr/slots
result:
 0: PF----  -1
 1: PF----  -1
 2: PF----  -1
 3: PF----  -1
 4: P-O-L-   0 Override Board Name,00A0,Override Manuf,cape-univ-hdmi

It also happens if the univ-hdmi cape is not installed.  What am doing wrong? 
Thanks, in advance...

ChrisB

cdsteinkuehler, 2016-09-29:
On 9/29/2016 2:47 PM, ChrisB wrote:
> kernel version - uname -r: 4.1.33-bone24-00149-gea9e337

Use the version of the universal cape that comes with RCN's kernel
builds (compiled into the kernel).  The numbering for various devices
(GPIO, serial ports, etc) has changed between kernel verisons.  Robert
tracks this and keeps the capes updated for the different kernel flavors.

-- 
Charles Steinkuehler
charles@steinkuehler.net

Biffco, 2016-09-29:
Use the version of the universal cape that comes with RCN's kernel
builds (compiled into the kernel).  The numbering for various devices
(GPIO, serial ports, etc) has changed between kernel verisons.  Robert
tracks this and keeps the capes updated for the different kernel flavors.

bolandrm, 2016-09-30:
I'm running into the same issue.    I've followed the steps in this readme, but i'm still not having any luck: https://github.com/beagleboard/bb.org-overlays

Would you be able to explain in a bit more detail what I need to do?  

```
config-pin -a P8.11 pruin
P8_11 pinmux file not found!
P8_11 overlay not found
Loading cape-universala overlay
bash: line 0: echo: write error: Invalid argument
Error loading device tree overlay file: cape-universala
```

```
# uname -r
4.4.22-ti-r48
```

```
[   80.158912] bone_capemgr bone_capemgr: part_number 'cape-universala', version 'N/A'
[   80.158944] bone_capemgr bone_capemgr: slot #4: override
[   80.158959] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[   80.158975] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,cape-universala'
[   80.217623] of_resolve_phandles: Could not find symbol 'gpio4'
[   80.223612] bone_capemgr bone_capemgr: slot #4: Failed to resolve tree
```

bolandrm, 2016-09-30:
Nevermind, i was a bit confused about what was going on.

It looks like https://github.com/beagleboard/bb.org-overlays doesn't container `cape-universalh`, so i used a combination of `cape-universal` and `cape-univ-hdmi`.

Biffco, 2016-09-30:
Config-pin attempts to load cape-universala but fails on the "Could not find symbol 'gpio4'" error.  The dts source file shows that the gpio(n+1) offset is present in the "a" file but not in the "n" file.
Manually installing the cape-universaln file works, but config-pin doesn't accept the "n" file and tries to install the "a" which produces the error.

I need to use config-pin for my development; any suggestions?

cdsteinkuehler, 2016-09-30:
On 9/30/2016 3:33 PM, ChrisB wrote:
> Config-pin attempts to load cape-universala but fails on the "Could not find
> symbol 'gpio4'" error. The dts source file shows that the gpio(n+1) offset is
> present in the "a" file but not in the [...]

Get a consistent set of device tree overlay files.

The files in this repository are consistent and all match the 3.8.13
kernel device tree numbering scheme.  If you continue to have problems
with the 4.x kernel, ask for help on the BeagleBoard Google Group or
open an issue with the proper repository for the 4.x kernel overlays:

https://github.com/beagleboard/bb.org-overlays

-- 
Charles Steinkuehler
charles@steinkuehler.net