cdsteinkuehler/beaglebone-universal-io #60: Error, Invalid mode: in

SmartFactory-github, 2019-10-17:
Hello!
I have issues using the config-pin -f .bbio command.

My bbio file, generated with the GUI is:
[DT-mk-probotix-bb.txt](https://github.com/cdsteinkuehler/beaglebone-universal-io/files/3741006/DT-mk-probotix-bb.txt)
But, as [RobertCNelson](https://github.com/machinekoder/BBIOConfig/issues/4) recommended, I removed the lines
```
overlay cape-universal
overlay cape-bone-iio
```
When I run
`sudo /opt/scripts/tools/version.sh`
I get
```
git:/opt/scripts/:[9599f8c47e18db5cdd36ee11355593ecdfe50979]
eeprom:[A335BNLTEIA05116BBBK3FEC]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[Machinekit Debian Image 2019-10-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-gf15b99f0b6]:[location: dd MBR]
kernel:[4.19.72-bone-rt-r39]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190922.0-0rcnee0~stretch+20190922]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[machinekit : machinekit adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.168353] remoteproc remoteproc0: wkup_m3 is available
[    1.283105] remoteproc remoteproc0: powering up wkup_m3
[    1.283124] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.285180] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.736328] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.747789] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

but when using `config-pin -f DT-mk-probotix-bb.bbio`, I obtain a `Invalid mode: in`
I attach my uEnv.txt configuration file.
[uEnv.txt](https://github.com/cdsteinkuehler/beaglebone-universal-io/files/3741017/uEnv.txt)

I plan to use this BBIO for Machinekit configuration for the Beaglebone Black

thanks in advance

regards,

Sergio Salinas

SmartFactory-github, 2019-10-17:
Now I am getting a 
```
P8_07 pinmux file not found!
Please verify your device tree file
```

RobertCNelson, 2019-10-18:
@SmartFactory-github remove the lines that dont' set anything..

```
P9_13 
```

Regards,

SmartFactory-github, 2019-10-19:
Hello @RobertCNelson !! thank you always for your quick assesment and response.
Well, I removed the lines that don't say anything, but still without success. Sometimes I get the `Invalid mode: in`, and after rebooting or compiling, I get `P8_07 pinmux file not found`.

So I am questioning my process, as it is the first time I try to define a **DT** and use it for the interface between machinekit and a **PBX-BB rev5.2** for the **BBB**. 
I need, once again, your assesment. I promise to make a public tutorial when this gets working.

I am using the **Machinekit** [image](https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#BBW.2FBBB_.28All_Revs.29_Machinekit) for beaglebone.

I am using two guides for my process, the one in the machinekit [blog](http://blog.machinekit.io/), and the one in the google [forum](https://groups.google.com/forum/#!topic/machinekit/SF9xA8sdpB0).

At first I used the BBIO GUI to generate the bbio file which defines the pin and configurations I need for MK. As I could not use properly the `config-pin -f <name of the file>` tool, I am now trying to edit directly the cape-universaln-00A0.dts file. 

This is where it gets confusing.
As I understood from the forum, I should remove all traces of the conflicting pins from the "exclusive-use" section, and from the interfaces (I2c, PWM, etc), but then how will the BBB know I need to use them?

As I undertood, then, supposing I have defined my .dts and I had compiled it, the next thing is to copy the .dtbo to `/lib/firware` and enable it in `/boot/uEnv.txt`.

After this, all I have rest to do, to define the pin assignment in a .hal file, and parameters in the .ini. right?

It has been troublesome as most of the documentation is for kernel 3.8 or earlier, and many things have changed for the recent version.

thanks in advance,

regards,

Sergio Salinas